mhvis/solar #2: Multiple inverters support

mhvis, 2016-05-18:
Small adjustments are needed for this.

chrleyer, 2016-06-06:
Please can you help to explain the small change how to pull information from two inverters. WHat do you mean by running the Inverter class two times?

mhvis, 2016-06-06:
(Excerpt from email conversation:)

I was planning to implement this myself. That is probably the easiest since I have the best understanding of the code. My idea for what I want to implement for multiple inverter support is the following:
- In solar_uploader.ini you can specify all inverters by their serial number (for my inverter that is 'DW413B8080').
- For each inverter you can specify the PVOutput system ID (and possibly API key) to use for uploading the data (in solar_uploader.ini).
- For inverters that have the same PVOutput system ID, the data will be accumulated or averaged (depending on the kind of data).

Any remarks/tips on this idea are welcome.

mhvis, 2016-06-06:
Specifying an inverter should also be possible using (static) IP

closed 2016-06-16 by mhvis