mhvis/solar #11: not getting data from solarlake TL-12000 to pvoutput

kevintieman, 2017-04-27:
Hi,

I'm tying to get data from the solarlake TL-12000 to pvoutput, however the script seems to be stuck at "INFO:samil:Receive buffer was not empty before a request" without ever connecting to pvoutput (see tcpdump below). The error message does not give much to go on, any ideas?

using Python 3.5.2 (also tried Python 2.7.12)

root@ubuntu:~# python3 solar/samil_upload.py
INFO:__main__:Configuration sections: ['DEFAULT']
INFO:samil:Searching for an inverter in the network
INFO:samil:Connected with inverter on address ('192.168.1.109', 1454)
INFO:__main__:Matched inverter 192.168.1.109 with configuration DEFAULT
INFO:__main__:Final systems: {'51489': (300, [192.168.1.109])}
INFO:__main__:Scheduled first upload
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
WARNING:__main__:Error occurred, restarting app in a minute: list index out of range
WARNING:samil:Error in keep alive thread: [Errno 9] Bad file descriptor
INFO:__main__:Configuration sections: ['DEFAULT']
INFO:samil:Searching for an inverter in the network
INFO:samil:Connected with inverter on address ('192.168.1.109', 1488)
INFO:__main__:Matched inverter 192.168.1.109 with configuration DEFAULT
INFO:__main__:Final systems: {'51489': (300, [192.168.1.109])}
INFO:__main__:Scheduled first upload
INFO:samil:Receive buffer was not empty before a request
INFO:samil:Receive buffer was not empty before a request
^CTraceback (most recent call last):
  File "solar/samil_upload.py", line 116, in <module>
    main(config)
  File "solar/samil_upload.py", line 105, in main
    scheduler.run()
  File "/usr/lib/python3.5/sched.py", line 145, in run
    delayfunc(time - now)
KeyboardInterrupt

root@ubuntu:/data/solar-master# tcpdump host 45.56.66.169
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on ens160, link-type EN10MB (Ethernet), capture size 262144 bytes

Thanks in advance!

Kevin

mhvis, 2017-04-30:
Hey Kevin, sorry for the kind of late response, my email notifications are not set up correctly (have to fix that after writing this post..). Thanks for the elaborate information. I suspect that the inverter sends some messages that are not in the exact same format as the script expects. The script has as far as I know not been tested with SolarLake inverters before, so your feedback is valuable to me :). It should however be possible to work on SolarLake TL inverters, as the protocol is the same.

I have lately actually been busy re-examining the protocol/messages that the inverter uses ([described here](https://github.com/mhvis/solar/wiki/Communication-protocol)), and rewriting the code in a different language (Go). With that, I fixed a couple of issues I came across. There is a small command-line application with which you can test if you can get generation/model info from the inverter via the new code [here](https://github.com/mhvis/samil/tree/master/cli) (binary to download is [here](https://github.com/mhvis/samil/releases/latest)). Can you run this and see if you get generation data? If it works, the issue has been fixed in the rewrite and you can simply wait for the new version to be finished (will be soon).

If it doesn't work, can you post the output of the binary? Also in that case, can you run tcpdump with the following expression and post that: `tcpdump 'tcp port 1200 or udp port 1300' -X` (if it is really long, maybe attach it as a txt or pcap file).

mhvis, 2017-05-05:
Hey Kevin, have you looked into my previous comment?

kevintieman, 2017-05-06:
Hi Maarten,

Apologies for not replying earlier, I do not actually own the inverter
(it's from a friend). I contacted my friend to access his system to test
this.

Will let you know!

Thanks again.

Kevin

mhvis, 2017-05-06:
Okay thank you, I was wondering if you had seen the comment.

kevintieman, 2017-05-06:
Hi,

Tried running the samil.go script (binary version):

bob@ubuntu:~$ ./samil-linux-amd64

searching for inverters

found inverter on address 192.168.1.109:3854

model info: {DeviceType:2 VARating:12000 FirmwareVersion:2.03
ModelName:SolarLake12K Manufacturer:SamilPower SerialNumber:T1212A9014
CommunicationVersion:2.01 OtherVersion:2.00 General:1}

data failed: expected response of length 54, got 68

PCAP data:

listening on ens160, link-type EN10MB (Ethernet), capture size 262144 bytes

20:01:45.147044 IP 192.168.1.109.3854 > 192.168.1.20.1200: Flags [S], seq
781010, win 365, options [mss 365,nop,nop,sackOK], length 0

        0x0000:  4500 0030 b70f 4000 8006 bfe6 c0a8 016d  E..0..@........m

        0x0010:  c0a8 0114 0f0e 04b0 000b ead2 0000 0000  ................

        0x0020:  7002 016d 038c 0000 0204 016d 0101 0402  p..m.......m....

20:01:45.147070 IP 192.168.1.20.1200 > 192.168.1.109.3854: Flags [R.], seq
0, ack 781011, win 0, length 0

        0x0000:  4500 0028 21da 4000 4006 9524 c0a8 0114  E..(!.@.@..$....

        0x0010:  c0a8 016d 04b0 0f0e 0000 0000 000b ead3  ...m............

        0x0020:  5014 0000 2d62 0000                      P...-b..

20:01:47.084894 IP 192.168.1.20.46416 > 255.255.255.255.1300: UDP, length 20

        0x0000:  4500 0030 8493 4000 4011 f46d c0a8 0114  E..0..@.@..m....

        0x0010:  ffff ffff b550 0514 001c c1e9 55aa 0040  .....P......U..@

        0x0020:  0200 0b49 2041 4d20 5345 5256 4552 043a  ...I.AM.SERVER.:

20:01:47.147052 IP 192.168.1.109.3854 > 192.168.1.20.1200: Flags [S], seq
781010, win 365, options [mss 365,nop,nop,sackOK], length 0

        0x0000:  4500 0030 b710 4000 8006 bfe5 c0a8 016d  E..0..@........m

        0x0010:  c0a8 0114 0f0e 04b0 000b ead2 0000 0000  ................

        0x0020:  7002 016d 038c 0000 0204 016d 0101 0402  p..m.......m....

20:01:47.147077 IP 192.168.1.20.1200 > 192.168.1.109.3854: Flags [S.], seq
4235788234, ack 781011, win 29200, options [mss 1460,nop,nop,sackOK],
length 0

        0x0000:  4500 0030 0000 4000 4006 b6f6 c0a8 0114  E..0..@.@.......

        0x0010:  c0a8 016d 04b0 0f0e fc78 ffca 000b ead3  ...m.....x......

        0x0020:  7012 7210 83f4 0000 0204 05b4 0101 0402  p.r.............

20:01:47.147274 IP 192.168.1.109.3854 > 192.168.1.20.1200: Flags [.], ack
1, win 365, length 0

        0x0000:  4500 0028 b711 4000 8006 bfec c0a8 016d  E..(..@........m

[... 103 lines cut ...]
        0x0000:  4500 0028 b716 4000 8006 bfe7 c0a8 016d  E..(..@........m

        0x0010:  c0a8 0114 0f0e 04b0 000b eb70 fc78 ffde  ...........p.x..

        0x0020:  5010 016d 2f04 0000 55aa 0182 0000       P..m/...U.....

20:01:47.457029 IP 192.168.1.109.3854 > 192.168.1.20.1200: Flags [F.], seq
158, ack 20, win 365, length 0

        0x0000:  4500 0028 b717 4000 8006 bfe6 c0a8 016d  E..(..@........m

        0x0010:  c0a8 0114 0f0e 04b0 000b eb70 fc78 ffde  ...........p.x..

        0x0020:  5011 016d 2f03 0000 55aa 0182 0000       P..m/...U.....

20:01:47.457051 IP 192.168.1.20.1200 > 192.168.1.109.3854: Flags [.], ack
159, win 29200, length 0

        0x0000:  4500 0028 23e0 4000 4006 931e c0a8 0114  E..(#.@.@.......

        0x0010:  c0a8 016d 04b0 0f0e fc78 ffde 000b eb71  ...m.....x.....q

        0x0020:  5010 7210 be5f 0000                      P.r.._..

Kevin

mhvis, 2017-05-06:
Awesome, very helpful :D. The message format seems indeed slightly different. I'll look into it and try to update the script soon.

mhvis, 2017-05-10:
Hi Kevin, it seems I am going to need more data for fixing the protocol for SolarLake. I tried to simulate the packet that your inverter sent with the Samil software (SolarPower Browser V3/4), but that gave some strange results (heatsink temperature of 500.5 degrees Celcius).

It would be cool if I could fix the script for SolarLake inverters, so if you could and want to help me with that, that would be really nice. If it is too hard to get the information, since you do not own the inverter, then too bad, but no problem ;)

If you plan to help me out and I manage to understand the protocol, I'll make sure to update the script in a timely manner so that you can actually use it. To help me you should do the following: run the official/proprietary Samil Power monitoring software, called SolarPower Browser. You can find it on this page, at the bottom: http://samilpower.com/service.php (should also run on Linux using Wine). You should check if it works for your inverter, i.e. you should be able see the current power generation values. If you know that it works, you should close it, FIRST start tcpdump and make it write to a file (you can use: `tcpdump 'tcp port 1200 or udp port 1300' -w dump.pcap`) and after that start SolarPower Browser again and make it connect to the inverter(s). Then you should make some screenshots of the data that is currently displayed in SolarPower Browser. If you could send the dump.pcap file and the screenshots, that would be awesome. You can attach them to the comment or send them to me directly (mail@maartenvisscher.nl).

Again, it would be nice if you could help me but no problem if that's not going to happen ;)

kevintieman, 2017-05-16:
Hi Maarten,

I will do this next time I'm at my friends place (friday).

Thanks!

Kevin

mhvis, 2017-07-12:
Hey Kevin! Any chance you might look back into this (if not, no problem)?

kevintieman, 2017-07-13:
Hi Maarten,

I have the following feedback (ben je trouwens nederlands?):

*The script gives an error:*

searching for inverters
found inverter on address 192.168.1.109:2667
model info: {DeviceType:2 VARating:12000 FirmwareVersion:2.03
ModelName:SolarLake12K Manufacturer:SamilPower SerialNumber:T1212A9014
CommunicationVersion:2.01 OtherVersion:2.00 General:1}
data failed: expected response of length 54, got 68

*TCP Dump on tcp port 1200:*

16:12:53.448534 IP Win10-VM.1200 > 192.168.1.109.2667: S
3731867324:3731867324(0) ack 781011 win 8192 <mss 1460,nop,nop,sackOK>
16:12:53.448783 IP 192.168.1.109.2667 > Win10-VM.1200: . ack 1 win 365
16:12:53.449331 IP Win10-VM.1200 > 192.168.1.109.2667: P 1:10(9) ack 1 win
65392
16:12:53.449587 IP 192.168.1.109.2667 > Win10-VM.1200: . ack 10 win 356
16:12:53.598486 IP 192.168.1.109.2667 > Win10-VM.1200: P 1:81(80) ack 10
win 365
16:12:53.598969 IP Win10-VM.1200 > 192.168.1.109.2667: P 10:19(9) ack 81
win 65312
16:12:53.599204 IP 192.168.1.109.2667 > Win10-VM.1200: . ack 19 win 356
16:12:53.698439 IP 192.168.1.109.2667 > Win10-VM.1200: P 81:158(77) ack 19
win 365
16:12:53.700260 IP Win10-VM.1200 > 192.168.1.109.2667: R 19:19(0) ack 158
win 0
16:13:03.683526 IP 192.168.1.109.2667 > Win10-VM.1200: P 81:158(77) ack 19
win 365
16:13:13.683363 IP 192.168.1.109.2667 > Win10-VM.1200: P 81:158(77) ack 19
win 365
16:13:33.693057 IP 192.168.1.109.2667 > Win10-VM.1200: F
4294967223:4294967223(0) ack 19 win 365
16:13:45.692868 IP 192.168.1.109.2670 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:47.692871 IP 192.168.1.109.2670 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:49.692840 IP 192.168.1.109.2670 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:51.692831 IP 192.168.1.109.2670 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:53.692809 IP 192.168.1.109.2670 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:55.692779 IP 192.168.1.109.2671 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:57.692746 IP 192.168.1.109.2671 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:13:59.692694 IP 192.168.1.109.2671 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>
16:14:01.692674 IP 192.168.1.109.2671 > Win10-VM.1200: S 781010:781010(0)
win 365 <mss 365,nop,nop,sackOK>

Kevin

mhvis, 2017-07-22:
Hi, the response is a bit late due to the holidays. I'm not able to find out how the protocol works with the information that I got (your last message has the same information as a previous message). I therefore can't make it work for this type of inverter, unless I got my hands on one. Sorry for that, and still thanks for your help.

PS: ja ik ben Nederlands inderdaad :)! Ik kom uit regio Eindhoven.

semonet, 2017-11-30:
Hey guys, 
I'm not sure if you are still interested but I just forked the project and made it work with my SolarLake TL-15000PM. Kevin, you can give it a try on your inverter if you want. https://github.com/semonet/solar
It's a quick port and it can probably be improved. 
Cheers,
Seb

mhvis, 2017-12-02:
Hey Seb! Very cool, looks neat :D!!

Thanks very much for sharing it :). I'll put a link to it in the README of this repo.

martiinezz, 2019-04-08:
Hi guys I have tried both scripts and they kinda fail. My output is:

searching for inverters
found inverter on address 192.168.15.139:3509
model info: {DeviceType:2 VARating:17000 FirmwareVersion:2.11 ModelName:SolarLake17K Manufacturer:SamilPower SerialNumber:T1712CC008 CommunicationVersion:2.11 OtherVersion:2.11 General:1}
data failed: expected response of length 54, got 68

Semonet script is unable to find this inverter. While mhvis code displays wrong information as for more data is returned by this 3 phase invertert.

I'm glad to help as I have access to this inverter everyday.

mhvis, 2019-04-08:
Your inverter uses a different communication protocol version just like the inverter from @kevintieman, as specified by the CommunicationVersion and General [values](https://github.com/mhvis/solar/wiki/Communication-protocol#modelversion-info). This version is currently not supported and I can't add the support for it myself as I don't have the inverter. The version by @semonet is for a different inverter type, it uses a different communication port and therefore won't find your inverter.

To have support added would mean that we need someone who has this inverter type and has the expertise needed to write the support for it. That boils down to examining the network packets that are sent using a tool like Wireshark or tcpdump and documenting the format of these packets.

martiinezz, 2019-04-08:
If this process is described somewhere with detailed explanation I can handle that. If not I can give you temporary access to the inverter via ssh.

mhvis, 2019-04-08:
In short it's like this: use the SolarPower Browser V3 software which you can download [here](http://www.samilpower.com/service.php) to connect to the inverter and have it display the values of interest. While it's running and during startup you'd capture the network packets which will contain these data values somewhere in the packets that get send between the program and the inverter. Then to find out which bytes in the packets indicate which type of value (voltage/current/temperature) you would compare the packet contents with the values displayed in the user interface to find the bytes in the packets that correspond with the values displayed in the interface. It may require capturing network traces from multiple moments to have some larger differences in operation values, e.g. during the day and at night.

You could try creating a network trace with e.g. Wireshark (pcap file) and at the same time creating screenshots of the program with timestamps, and I'd gladly look into them to see if I can figure it out :smile:. I can't promise that I'll be able to though, maybe it requires more data. When you send a trace, watch out that it'll by default contain all your network traffic and may contain sensitive data, although usually all sensitive data is encrypted using HTTPS.

If you want to try it yourself, you can use the following filter in Wireshark: `udp.port==1300||tcp.port==1200`. When you start SolarPower Browser V3 with this filter you should start seeing the relevant packets appearing. The format of the bytes that are in each packet is documented [here](https://github.com/mhvis/solar/wiki/Communication-protocol). The specific data types mentioned are obviously not correct for your data packets, as we're trying to figure that out right now :grinning:

mhvis, 2019-04-08:
Temporary SSH access would probably work as well, then I would forward port 1200 and then I should be able to run SolarPower Browser locally and connect to the inverter.

martiinezz, 2019-04-09:
With this filer I only see broadcast traffic (I'm a server) nothing else :(

mhvis, 2019-04-09:
I did only test the broadcast filter, not the TCP filter, however it looks correct to me when looking at the [manual](https://www.wireshark.org/docs/wsug_html_chunked/ChWorkBuildDisplayFilterSection.html). Did you actually see the inverter appearing in the interface? If so, you could maybe try only the TCP filter `tcp.port==1200` or filter on the inverter IP address or no filter at all.

martiinezz, 2019-04-09:
This is how samil.py reacts. Having IP filter in wireshark didn't help. Do I need to run it as administrator?

DEBUG:__main__:Binding TCP socket to :1200
DEBUG:__main__:Binding UDP socket to :0
INFO:__main__:Searching for an inverter in the network
DEBUG:__main__:Broadcasting server existence
INFO:__main__:Connected with inverter on address ('192.168.15.139', 3560)
DEBUG:__main__:Request: b'U\xaa\x01\x02\x02\x00\x00\x01\x04'
DEBUG:__main__:Response: (b'\x01\x82\x00', b'\x01^\x16\xf6\x00C\x00L\x00\x01\x00\x00\x02U\x00\x00\x00(\x10\xda\x00\x01\x01\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11\xac\x00\x00\x01\xe0\x00;\t`\x13\x86\x00=\t\x83\x13\x86\x00<\tZ\x13\x86', b"\x0b'")
DEBUG:__main__:Current values: {'grid_current': 0.0, 'pv1_current': 7.6, 'energy_total': 5.9, 'pv1_voltage': 587.8, 'internal_temp': 35.0, 'total_operation_hours': 597, 'pv2_input_power': 0, 'energy_today': 0.4, 'pv2_current': 0.1, 'grid_voltage': 452.4, 'pv1_input_power': 0, 'grid_frequency': 0.0, 'pv2_voltage': 6.7, 'output_power': 480, 'operating_mode': 'wait'}
DEBUG:__main__:Request: b'U\xaa\x01\x03\x02\x00\x00\x01\x05'
DEBUG:__main__:Response: (b'\x01\x83\x00', b'2 170002.11\x00SolarLake17K    SamilPower      T1712CC008\x00\x00\x00\x00\x00\x002.11\x002.11\x001', b'\x11<')
INFO:__main__:Model info: (b'\x01\x83\x00', b'2 170002.11\x00SolarLake17K    SamilPower      T1712CC008\x00\x00\x00\x00\x00\x002.11\x002.11\x001', b'\x11<')
DEBUG:__main__:Request: b'U\xaa\x01\x00\x02\x00\x00\x01\x02'
DEBUG:__main__:Response: (b'\x01\x80\x00', b'\x00\x01\x02\x04\x05\x07\x08\t\n\x0b\x0c\x11\x17\x18\x19\x1a\x1b\x1c\x1d\x1e!"\'(/123QRSqrs', b'\x06o')
DEBUG:__main__:Request: b'U\xaa\x01\t\x02\x00\x00\x01\x0b'
DEBUG:__main__:Response: (b'', b'', b'')

martiinezz, 2019-04-09:
[image]

martiinezz, 2019-04-09:
Looks like PV1 voltage and PV1 current are correct as is PV2 voltage and PV2 current.

mhvis, 2019-04-09:
> Having IP filter in wireshark didn't help. Do I need to run it as administrator?

That might help, or maybe the selected interface is incorrect. I don't have time today to look into it but since `samil.py` doesn't crash and the values look very similar, maybe the problem is somewhere else. I'm going to examine it somewhere this week. If you want to try for yourself, you could try to find out the exact problem by disabling the error suppression, by replacing these bottom lines of `samil_upload.py`:

```python
    # Restart on errors
    import socket
    while True:
        try:
            main(config)
        except (socket.error, IndexError) as err:
            logger.warning('Error occurred, restarting app in a minute: %s', err)
            time.sleep(60)
```

with:

```python
    while True:
        main(config)
```

This might give a useful exception message

martiinezz, 2019-04-09:
I was able to do wireshark capture.

[image]

[inverter.pcap.gz](https://github.com/mhvis/solar/files/3059342/inverter.pcap.gz)

martiinezz, 2019-04-09:
Cool

mhvis, 2019-04-11:
I'm using the opportunity to do a rewrite that I wanted to do already for some time, to get rid of the scheduler which can be confusing and is not necessary and will also keep the connection open unnecessarily. Since you're using EmonCMS I assume you've modified the script, so then you'll most likely have to modify it again, sorry for that. You can check the new readme [here](https://github.com/mhvis/solar/tree/v2) to see what I have in mind for the rewrite, I expect to finish it tomorrow with support for your inverter types.

martiinezz, 2019-04-13:
Hi,
I didn't modify the script itself - I'm running listener that uses samil.py as module and responds with JSON with all the values your script provides. Then I'm just picking stuff I want and mainly because my main code is python2 while yours is python3. Now as I replaced 2 inverters with one 3 phase I might be able to jump to python3 as the other inverter was only having rs485 python code available (and now it's gone).

martiinezz, 2019-04-13:
#!/usr/bin/env python3

from http.server import BaseHTTPRequestHandler, HTTPServer
from urllib.parse import urlparse
import json
import samil

with samil.InverterListener() as listener:
    inverter = listener.connect()

class RequestHandler(BaseHTTPRequestHandler):
    def do_GET(self):
        parsed_path = urlparse(self.path)
        self.send_response(200)
        self.end_headers()
        values = inverter.request_values()
        self.wfile.write(json.dumps(values).encode())
        return

if __name__ == '__main__':
    server = HTTPServer(('', 8000), RequestHandler)
    print('Starting server port 8000')
    server.serve_forever()

mhvis, 2019-04-13:
Perfect :smile:, then you'll probably only have to rename some things.

I did find out how the current/voltage packet format works, thanks to your network packets! I updated the [doc page](https://github.com/mhvis/solar/wiki/Communication-protocol#status-data-format) with the new findings and next step is to update the source code so that it adheres to the new doc. That should hopefully make the script work for all applicable inverters that work with the SolarPower Browser tool (I'm basically just trying to mimic the SolarPower Browser tool)

martiinezz, 2019-04-13:
I'm glad I could help to my benefit too ;)

mhvis, 2019-04-14:
I have an updated version and I'm very curious to see how it performs on your inverter :smile:. It is [here](https://github.com/mhvis/solar/tree/v2) and you can test it using `python3 -m solar`. I'm still working on it to add the possibility of requesting history data and to restore PVOutput uploading which is not yet implemented. Some notable differences with the previous version are that almost all status values are now returned as Decimal instead of float, one of the reasons being that the Decimal class preserves significance. Model information is added. Also I have removed the keep alive mechanism. That might mean that you have to accommodate for that yourself. The inverter closes the connection automatically when there's no activity after some time, I believe around 20 seconds. That means that you have to make sure that some data is requested at least every 20 seconds to keep the connection open, or you'd have to reestablish the connection each time.

mhvis, 2019-04-14:
I could actually put the keep alive mechanism back into the script if you would like that, now that I think of it, just by extending the current `Inverter` class.

martiinezz, 2019-04-14:
INFO:Searching for inverter
INFO:Connected with inverter on address ('192.168.15.139', 4425)
{'communication_version': '2.11',
 'device_type': 'Three-phase inverter',
 'firmware_version': '2.11',
 'general': '1',
 'manufacturer': 'SamilPower',
 'model_name': 'SolarLake17K',
 'other_version': '2.11',
 'serial_number': 'T1712CC008',
 'va_rating': '17000'}
{'energy_today': Decimal('18.20'),
 'energy_total': Decimal('154.6'),
 'grid_current': Decimal('5.9'),
 'grid_current_s_phase': Decimal('6.1'),
 'grid_current_t_phase': Decimal('6.0'),
 'grid_frequency': Decimal('50.01'),
 'grid_frequency_s_phase': Decimal('50.01'),
 'grid_frequency_t_phase': Decimal('50.01'),
 'grid_voltage': Decimal('241.1'),
 'grid_voltage_s_phase': Decimal('247.1'),
 'grid_voltage_t_phase': Decimal('246.9'),
 'heatsink_temperature': Decimal('58.0'),
 'internal_temperature': Decimal('35.0'),
 'operating_mode': 'Normal',
 'output_power': Decimal('4405'),
 'pv1_current': Decimal('8.1'),
 'pv1_input_power': Decimal('4636'),
 'pv1_voltage': Decimal('568.3'),
 'pv2_current': Decimal('0.1'),
 'pv2_input_power': Decimal('1'),
 'pv2_voltage': Decimal('6.8'),
 'total_operation_time': 110}

Looks awesome!!!

mhvis, 2019-04-14:
Nice, you can use this as-is, I'm still going to update some things like history data and PVOutput, but probably nothing that will affect these status values. You'll have to rename some things to get something like this:
```python
import solar

with solar.InverterListener() as listener:
    inverter = listener.accept_inverter()
    print(inverter.model()) # Model info
    print(inverter.status()) # Status data
```

Also the keep alive might be a problem still if the requests are interrupted for some reason, but you'll have to see for yourself.

martiinezz, 2019-04-15:
I'm getting values like Decimal('35.0')

mhvis, 2019-04-15:
Yes, this is due to that I changed the values from `float` to `Decimal` type because `Decimal` doesn't suffer from precision issues and preserves significance. You can convert between the types using `float(value)` or `str(value)`.

```python
>>> from decimal import Decimal
>>> d = Decimal('1.20')
>>> d
Decimal('1.20')
>>> float(d)
1.2
>>> str(d)
'1.20'
```

martiinezz, 2019-04-16:
Ok so I have changed how data is returned in methods and it works as I want it to :) thanks.

dzban1976, 2019-10-27:
SolarLake 7000TL-PM possible reading?
Reasult scan tpcdump:

> 11:19:14.009901 IP (tos 0x0, ttl 64, id 1358, offset 0, flags [none], proto UDP                                                                                                                      (17), length 348)
>     192.168.1.145.mdns > 224.0.0.251.mdns: [udp sum ok] 0*- [0q] 4/0/0 _http._tc                                                                                                                     p.local. PTR SAMILPOWER (111C4B)._http._tcp.local., SAMILPOWER (111C4B)._http._t                                                                                                                     cp.local. (Cache flush) TXT "Vendor=MXCHIP" "Version=00000000.006" "Socket1_Type                                                                                                                     =UDP unicast" "Socket1_LPort=1500" "Socket1_RPort=1400" "Socket1_Raddr=www.v-ipl                                                                                                                     ant.com" "", SAMILPOWER (111C4B)._http._tcp.local. (Cache flush) SRV EMW3280_111                                                                                                                     C4B.local.:80 0 0, EMW3280_111C4B.local. (Cache flush) A 192.168.1.145 (320)
>         0x0000:  4500 015c 054e 0000 4011 d10e c0a8 0191  E..\.N..@.......
>         0x0010:  e000 00fb 14e9 14e9 0148 e63a 0000 8400  .........H.:....
>         0x0020:  0000 0004 0000 0000 055f 6874 7470 045f  ........._http._
>         0x0030:  7463 7005 6c6f 6361 6c00 000c 0001 0000  tcp.local.......
>         0x0040:  012c 0016 1353 414d 494c 504f 5745 5220  .,...SAMILPOWER.
>         0x0050:  2831 3131 4334 4229 c00c 1353 414d 494c  (111C4B)...SAMIL
>         0x0060:  504f 5745 5220 2831 3131 4334 4229 c00c  POWER.(111C4B)..
>         0x0070:  0010 8001 0000 012c 0082 0d56 656e 646f  .......,...Vendo
>         0x0080:  723d 4d58 4348 4950 1456 6572 7369 6f6e  r=MXCHIP.Version
>         0x0090:  3d30 3030 3030 3030 302e 3030 3618 536f  =00000000.006.So
>         0x00a0:  636b 6574 315f 5479 7065 3d55 4450 2075  cket1_Type=UDP.u
>         0x00b0:  6e69 6361 7374 1253 6f63 6b65 7431 5f4c  nicast.Socket1_L
>         0x00c0:  506f 7274 3d31 3530 3012 536f 636b 6574  Port=1500.Socket
>         0x00d0:  315f 5250 6f72 743d 3134 3030 1e53 6f63  1_RPort=1400.Soc
>         0x00e0:  6b65 7431 5f52 6164 6472 3d77 7777 2e76  ket1_Raddr=www.v
>         0x00f0:  2d69 706c 616e 742e 636f 6d00 1353 414d  -iplant.com..SAM
>         0x0100:  494c 504f 5745 5220 2831 3131 4334 4229  ILPOWER.(111C4B)
>         0x0110:  c00c 0021 8001 0000 012c 001c 0000 0000  ...!.....,......
>         0x0120:  0050 0e45 4d57 3332 3830 5f31 3131 4334  .P.EMW3280_111C4
>         0x0130:  4205 6c6f 6361 6c00 0e45 4d57 3332 3830  B.local..EMW3280
>         0x0140:  5f31 3131 4334 4205 6c6f 6361 6c00 0001  _111C4B.local...
>         0x0150:  8001 0000 012c 0004 c0a8 0191            .....,......

interesting are the ports and the page where it sends data

mhvis, 2019-10-27:
This is not applicable here, this is a different inverter type which uses different ports.

mhvis, 2021-01-08:
SolarLake TL inverters should now work more reliably with the new version.

closed 2021-01-08 by mhvis