mhvis/solar #12: SolarRiver 5200TL-D: Not uploading, no inverter has operating mode normal

larsomen, 2017-12-16:
I just tried the script but it doesn't seem to be working for my SolarRiver 5200TL-D.

I get the following output when i run the script:

INFO:__main__:Configuration sections: ['DEFAULT']
INFO:samil:Searching for an inverter in the network
INFO:samil:Connected with inverter on address ('192.168.1.9', 4106)
INFO:__main__:Matched inverter 192.168.1.9 with configuration DEFAULT
INFO:__main__:Final systems: {'52122': (60, [192.168.1.9])}
INFO:__main__:Scheduled first upload
INFO:__main__:Not uploading, no inverter has operating mode normal

It seems that the values variable is empty ( logger.info(values) gave no output)
I speak dutch if that is easier :)

mhvis, 2017-12-16:
Hi :)! I'll respond in English so that others can read it as well.

Your output seems reasonably okay :). The message is a bit misleading, but it simply says that the inverter reports an operating mode which is different than 'normal'. That happens e.g. when there is no sunlight and the inverter is therefore not running. It is pretty dark here at the moment, so if you were running it just now than that could be the case ;). You can retry it tomorrow while there is sunlight.

You can also move `logger.info(values)` to directly above the line which says `# Filter systems with normal operating mode`. That should give you a non-empty `values` variable.

Good luck with your inverter setup!

larsomen, 2017-12-16:
Thanks! Will try it tomorrow!

larsomen, 2017-12-18:
It seems to be working great, thanks for the help!

mhvis, 2017-12-18:
Cool, thanks for the feedback :)!

closed 2019-04-14 by mhvis