mhvis/solar #13: Error occurred, restarting app in a minute: [Errno 98] Address already in use

fbijma2, 2019-10-23:
Hi Maarten,
I keep getting adddress already in use. I have multiple interfaces in my system and would like te script to search at only 192.168.50.2 interface. But it does not seam to work:

INFO:__main__:Configuration sections: ['DEFAULT']
WARNING:__main__:Error occurred, restarting app in a minute: [Errno 98] Address already in use
^CTraceback (most recent call last):
  File "samil_upload.py", line 116, in <module>
    main(config)
  File "samil_upload.py", line 73, in main
    with samil.InverterListener(interface_ip) as listener:
  File "/home/fbijma/solar/samil.py", line 31, in __init__
    self.server.bind((interface_ip, listen_port))
OSError: [Errno 98] Address already in use

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "samil_upload.py", line 119, in <module>
    time.sleep(60)
KeyboardInterrupt

this is my ini:
[DEFAULT]
# Number of minutes between uploads
Status interval = 5
Interface IP = 192.168.50.2

API key = **********************************
System ID = 70928

I assumed the API key is the hex key.
Hope you can help. Maybe missing some python modules?
Regards Frans

mhvis, 2019-10-24:
To clarify, do you mean that there are multiple Samil Power inverters in the network and you would like to only upload the data of the inverter at IP address 192.168.50.2?

mhvis, 2019-10-24:
Assuming you meant as above, the following configuration should probably work:

```
[DEFAULT]
Status interval = 5

API key = ***
System ID = 70928

[System1]
IP address = 192.168.50.2
```

Admittedly, this is not very intuitive, see [here](https://github.com/mhvis/solar#multiple-inverters-configuration) for an explanation.

fbijma2, 2019-10-25:
I have read all that, but 192.168.50.2 is the interface of my NUC that is
communicating to my network. 192.168.50.7 is the address of my convertor.

mhvis, 2019-10-25:
Okay interesting :+1:. There seems to be some other program which is using the server listening port, which is 1200. You could try to find out what that is using a command like `lsof -i :1200`. If you have another instance of the script running (on the same interface or all interfaces) than that is the problem. If nothing shows up for port 1200 it might have something to do with the `SO_REUSEADDR` option.

mhvis, 2019-10-25:
You might need to do `sudo lsof -i :1200` to see all processes and look for `LISTEN`

mhvis, 2021-01-08:
If you're still interested in this, you could try the new version.

closed 2021-01-08 by mhvis