mhvis/solar #14: SolarLake 7000TL-PM - data download

dzban1976, 2019-10-26:
Unfortunately, the producer's website is already dead...
I tried your script but my ports are 1500 and 1400 udp.
Port TCP 80 is also available but only for simple network configuration.
Below is the result from tcpdump.
Maybe you have any tips?
Thx.
```
16:28:05.302579 IP (tos 0x0, ttl 64, id 2502, offset 0, flags [none], proto UDP (17), length 348)
    192.168.1.140.mdns > 224.0.0.251.mdns: [udp sum ok] 0*- [0q] 4/0/0 _http._tcp.local. PTR SAMILPOWER (111C4B)._http._tcp.local.,                                     SAMILPOWER (111C4B)._http._tcp.local. (Cache flush) TXT "Vendor=MXCHIP" "Version=00000000.006" "Socket1_Type=UDP unicast" "Socket1_L                                    Port=1500" "Socket1_RPort=1400" "Socket1_Raddr=www.v-iplant.com" "", SAMILPOWER (111C4B)._http._tcp.local. (Cache flush) SRV EMW3280                                    _111C4B.local.:80 0 0, EMW3280_111C4B.local. (Cache flush) A 192.168.1.140 (320)
        0x0000:  4500 015c 09c6 0000 4011 cc9b c0a8 018c  E..\....@.......
        0x0010:  e000 00fb 14e9 14e9 0148 e644 0000 8400  .........H.D....
        0x0020:  0000 0004 0000 0000 055f 6874 7470 045f  ........._http._
        0x0030:  7463 7005 6c6f 6361 6c00 000c 0001 0000  tcp.local.......
        0x0040:  012c 0016 1353 414d 494c 504f 5745 5220  .,...SAMILPOWER.
        0x0050:  2831 3131 4334 4229 c00c 1353 414d 494c  (111C4B)...SAMIL
        0x0060:  504f 5745 5220 2831 3131 4334 4229 c00c  POWER.(111C4B)..
        0x0070:  0010 8001 0000 012c 0082 0d56 656e 646f  .......,...Vendo
        0x0080:  723d 4d58 4348 4950 1456 6572 7369 6f6e  r=MXCHIP.Version
        0x0090:  3d30 3030 3030 3030 302e 3030 3618 536f  =00000000.006.So
        0x00a0:  636b 6574 315f 5479 7065 3d55 4450 2075  cket1_Type=UDP.u
        0x00b0:  6e69 6361 7374 1253 6f63 6b65 7431 5f4c  nicast.Socket1_L
        0x00c0:  506f 7274 3d31 3530 3012 536f 636b 6574  Port=1500.Socket
        0x00d0:  315f 5250 6f72 743d 3134 3030 1e53 6f63  1_RPort=1400.Soc
        0x00e0:  6b65 7431 5f52 6164 6472 3d77 7777 2e76  ket1_Raddr=www.v
        0x00f0:  2d69 706c 616e 742e 636f 6d00 1353 414d  -iplant.com..SAM
        0x0100:  494c 504f 5745 5220 2831 3131 4334 4229  ILPOWER.(111C4B)
        0x0110:  c00c 0021 8001 0000 012c 001c 0000 0000  ...!.....,......
        0x0120:  0050 0e45 4d57 3332 3830 5f31 3131 4334  .P.EMW3280_111C4
        0x0130:  4205 6c6f 6361 6c00 0e45 4d57 3332 3830  B.local..EMW3280
        0x0140:  5f31 3131 4334 4205 6c6f 6361 6c00 0001  _111C4B.local...
        0x0150:  8001 0000 012c 0004 c0a8 018c            .....,......

```

mhvis, 2019-10-26:
If you need a script for SolarLake TL-PM series you could try this: https://github.com/semonet/solar. But it doesn't use port 1400 and 1500, so I don't think it applies to you. Other than that I don't really understand your question.

dzban1976, 2019-10-26:
https://github.com/semonet/solar - the script cannot detect the inverter.

> Other than that I don't really understand your question.
You just gave me an answer :)
The topic can be closed and there is no way for my inverter...

mhvis, 2019-10-26:
The tcpdump doesn't look familiar to me, I don't know how to interpret it and what it is from. Also I've not heard of v-iplant before.

Sorry that I can't help you

dzban1976, 2019-10-26:
v-iplant.com - is the page where data from the inverter was sent. (China cloud)
it is possible that the inverter was imported directly from China..
Inverters for the European market have different firmware.
But it's just a theory :).

Nytrm, 2020-03-20:
Maybe [this pdf](http://www.radio-active.net.au/images/files/Samil%20Inverter.pdf) might help, not sure if it is using the same commands.

mhvis, 2020-03-20:
The PDF is not loading for me.

Nytrm, 2020-03-20:
Try[ this link](https://4html.net/PDFViewer/#/source/Samil%20Inverter.pdf) pdf to html converted

mhvis, 2020-03-20:
Ah yes I've seen that PDF before, if I recall correctly I wasn't able to get a lot of useful information from it because it seems to be for a different inverter type than the one that I (*my parents) have. It does look like it's a good reference but the context is missing (mostly which inverters are actually described in the document). It could of course very well be for the interter that OP is using.

I have a similar document here: https://github.com/mhvis/solar/wiki/Communication-protocol

closed 2020-12-17 by mhvis