mhvis/solar #16: River 3400TL-D Works

Nytrm, 2020-03-26:
SamilPower River 3400TL-D werkt prima, keurig de data naar pvoutput.org

Handig aangezien het interface van de website niet meer werkt voor dit model.

mhvis, 2020-03-26:
Nice :smile: ik gebruik zelf nu niet meer PVOutput dus ook niet meer dit script, maar ik kijk nog wel naar bug reports of feature requests. Heb de readme net nog een beetje aangepast.

alochem, 2020-06-15:
Ik gebruik al jaaaaren een oude laptop die op zolder staat om de data via Samilpowerbrowser.exe uit de SamilPower te halen. Die laptop begint wat te kraken, dus op zoek naar een alternatief.
Daarbij kwam ik dit fijne script tegen. Dat werkt als een speer. Ik heb een Raspberry draaien PiHole erop, ik dacht dat het daar maar eens op moest gaan draaien.
Script (samilupload2) aangepast en naar /etc/init.d gekopieerd en het automatisch opstarten enabled. Maar helaas, geen upload na rebooten. 
Wat zou er mis kunnen zijn?
Dat gaat prima, alleen krijg ik het niet voor elkaar om het script automatisch te laten starten.

mhvis, 2020-06-15:
Ik neem aan dat je `WORKDIR` hebt aangepast in `samilupload2`. Je kan het volgende proberen:

```
$ /etc/init.d/samilupload2 restart
$ /etc/init.d/samilupload2 status
$ sudo service samilupload status
```
Wat is de output hiervan?

alochem, 2020-06-15:
Ik heb samilupload2 vervangen door samilupload, die krijg ik wel aan de gang. Het lijkt erop dat samilupload2 niet als service wil draaien bij mij.
Het maakt me niet zoveel uit of ik nu v1 of v2 draai, het draait nu.

Bedankt voor de snelle reactie!

mhvis, 2020-06-15:
`samilupload2` is niet een nieuwere versie 2 van dat script maar een andere manier, zoals uitgelegd in de [readme](https://github.com/mhvis/solar/tree/master/sysv). Het is sowieso beter om de standaard `samilupload` te gebruiken als die werkt.

Mocht je overigens andere functionaliteit nodig hebben in plaats van PVOutput uploading, ik ben toevallig nu (deels op verzoek) bezig met een nieuwe [v2](https://github.com/mhvis/solar/tree/v2). Eventueel zijn feature requests welkom.

alochem, 2020-06-15:
Dat ziet er ook goed uit.
Voorlopig hou ik het even hierbij, ben al blij dat ik de output bij PVOutput krijg buiten m'n oude laptop om. 
Zal er binnenkort eens naar kijken!

alochem, 2020-07-27:
Het script loopt prima, alleen om de zoveel tijd (kan een dag zijn, maar ook enkele dagen, wordt m'n inverter niet meer gevonden. Ik moet dan het script herstarten, dan werkt ie weer. Mijn broer met exact dezelfde omvormer (TL3400-D) heeft het probleem niet. 
Ik laat nu de service elke 24 uur herstarten. Eerst om 01.00 uur 's nachts, maar dat liep ook niet helemaal goed. Nu heb ik 'm op 9 uur gezet. Maar het lijkt me niet dat dit nodig is. Enig idee?

mhvis, 2020-07-27:
Het script zou bij fouten zoals verlies van verbinding automatisch moeten herstarten. Je kan het init.d script instellen zo dat de output gelogd wordt, door [deze regel](https://github.com/mhvis/solar/blob/51db77c1c11e1f0ed4c6d8651bbafc375547e2e4/sysv/samilupload#L22) te uncommenten. En dan kan in het logbestand gekeken worden wat het probleem is.

alochem, 2020-07-27:
Logging heb ik vanaf het begin aan gehad, zie bijlage.

[samilupload.log](https://github.com/mhvis/solar/files/4982484/samilupload.log)

alochem, 2020-09-10:
Heb je toevallig nog kans gezien naar de log te kijken om te zien wat er evt. fout gaat?

mhvis, 2020-09-10:
Ik kon uit de log niet halen waar het probleem zit. Ik ben er nog steeds niet aan toegekomen om v2 helemaal af te maken ook al is die zo goed als af. Als die af is kun je die proberen, goede kans dat die niet hetzelfde probleem heeft.

WendelVanRheenen, 2020-12-17:
Hi @mhvis , awesome project! werkt fantastisch met m'n Samil Solar River 3400TL-D. Als ik de **samil monitor** start geeft ie keurig de info terug. Krijg alleen die MQTT niet goed aan de gang. De help geeft aan dat ik bijvoorbeeld niet de interval tijd kan aangeven. Maar belangrijker nog, zie niet in terug wat de topic is waarop ik moet subscriben aan de andere kant van de brooker. Heb Mosquitto als broker op m'n pi gezet en als ik daarmee verbind met "#" als topic krijg ik alleen "hihi"  terug.

Dus ben even wat gaan rondkijken in de code en zag [hier dat je het formaat](https://github.com/mhvis/solar/blob/34c611d8c7cc4b909263bcc71968ce9741b27d74/samil/cli.py#L176) "inverter/[seririe nummer]/status gebruikt. Als ik daarop abboneer bij Mosquitto blijft het ook stil over de lijn. 

Heb je enig idee/suggestie wat het zou kunnen zijn?

mhvis, 2021-01-07:
Hoi @alochem, inmiddels een paar maanden verder maar ik heb nu eindelijk v2 gereleased, mocht je nog tegen hetzelfde probleem aanlopen kan je die proberen (zie de hoofdpagina).

alochem, 2021-01-23:
Hallo Maarten, ik ben gisteren met v2 van je programma bezig geweest. Paar vraagjes als het mag.
Ik heb een service aangemaakt welke op zich goed werkt. Bij het opstarten wordt de omvormer opgezocht, gegevens eruit gehaald en naar pvoutput gestuurd. Dat werkt goed. Daarna is de status van de service echter 'inactief' om daar niet meer uit te komen.
Hoe krijg ik de service zodanig ingesteld dat 'ie constant (om de 5 minuten) de gegevens uit de omvormer trekt en naar pvoutput stuurt?
Vooralsnog heb ik een regel opgenomen in de crontab welke elke 5 minuten ~/.local/bin/samil pvoutput <id> <api-key> uitvoert. Dat werkt ook, maar de service moet dat toch ook kunnen doen? Waar heb je deze anders voor?

mhvis, 2021-01-23:
Cron is een goede manier om het script elke 5 minuten te laten uploaden, je hebt dan geen service nodig, dat is hoe ik het nu zelf ook heb ingesteld. Als je liever een SystemD service gebruikt i.p.v. cron heb je 2 opties:

* Je kan het script zo starten dat het blijft draaien en elke 5 minuten uploadt i.p.v. dat het 1 keer uploadt en dan stopt. Dat kan door de optie `--interval 5` mee te geven. Ik raad deze optie alleen minder aan omdat het script dan continu verbonden blijft met de inverter en dat kan problemen geven als de verbinding een keer verbroken wordt o.i.d.
* Je kan een SystemD timer gebruiken om de service elke 5 minuten te starten, dit is vergelijkbaar met cron. Zie bijvoorbeeld [hier](https://wiki.archlinux.org/index.php/Systemd/Timers).

Om de data in Domoticz te krijgen hebben anderen succesvol gebruik gemaakt van MQTT, alleen vereist dit best wat installeerwerk omdat je Mosquitto en Node-RED moet installeren, zie [hier](https://www.domoticz.com/wiki/MQTT). Een plugin voor Domoticz zou wellicht makkelijker zijn maar ik heb geen ervaring hiermee.

alochem, 2021-01-23:
Hoi Maarten, dank voor je uitgebreide antwoord! Ik ga eens naar je linkjes kijken.