mhvis/solar #18: MQTT interval+topic

mhvis, 2020-12-17:
Hi @mhvis , awesome project! werkt fantastisch met m'n Samil Solar River 3400TL-D. Als ik de **samil monitor** start geeft ie keurig de info terug. Krijg alleen die MQTT niet goed aan de gang. De help geeft aan dat ik bijvoorbeeld niet de interval tijd kan aangeven. Maar belangrijker nog, zie niet in terug wat de topic is waarop ik moet subscriben aan de andere kant van de brooker. Heb Mosquitto als broker op m'n pi gezet en als ik daarmee verbind met "#" als topic krijg ik alleen "hihi"  terug.

Dus ben even wat gaan rondkijken in de code en zag [hier dat je het formaat](https://github.com/mhvis/solar/blob/34c611d8c7cc4b909263bcc71968ce9741b27d74/samil/cli.py#L176) "_inverter/[seririe nummer]/status_" gebruikt. Als ik daarop abboneer bij Mosquitto blijft het ook stil over de lijn. 

Heb je enig idee/suggestie wat het zou kunnen zijn?

_Originally posted by @WendelVanRheenen in https://github.com/mhvis/solar/issues/16#issuecomment-747458879_

mhvis, 2020-12-17:
Als je `samil mqtt --help` uitvoert krijg je de uitleg van het commando. Als je `samil mqtt` uitvoert zonder extra opties probeert het script te verbinden met een broker op adres localhost:1883, zonder authenticatie. Als het verbinden met broker en inverter lukt print het script de topicnaam.

Hier is de output van het commando `samil mqtt --help`:
```
$ samil mqtt --help
Usage: samil mqtt [OPTIONS]

  Publish inverter data to an MQTT broker.

  The default topic format is inverter/<serial number>/status, e.g.
  inverter/DW413B8080/status. The message value is a JSON object with all
  status data from the inverter. Example message value:

      {"operation_mode":"Normal","total_operation_time":45,
      "pv1_input_power":2822.0,"pv2_input_power":0.0,"pv1_voltage":586.5,
      "pv2_voltage":6.7,"pv1_current":4.8,"pv2_current":0.1,
      "output_power":2589.0,"energy_today":21.2,"energy_total":77.0,
      "grid_voltage":242.6,"grid_current":3.6,"grid_frequency":50.01,
      "internal_temperature":35.0}

Options:
  -n, --inverters INTEGER  Number of inverters.  [default: 1]
  -i, --interval FLOAT     Interval between status messages in seconds.
                           [default: 10.0]

  -h, --host TEXT          MQTT broker hostname or IP.  [default: localhost]
  -p, --port INTEGER       MQTT broker port.  [default: 1883]
  --client-id TEXT         MQTT client ID. If not provided, one will be
                           randomly generated.

  --tls                    Enable MQTT SSL/TLS support.
  --username TEXT          MQTT username.
  --password TEXT          MQTT password.
  --topic-prefix TEXT      MQTT topic prefix.  [default: inverter]
  --interface TEXT         IP address of local network interface to bind to.
  --help                   Show this message and exit.
```

WendelVanRheenen, 2020-12-17:
Awesome, gaat t meteen proberen; bedankt @mhvis  voor de snelle reactie!

hmm, krijg t gevoel een andere versie te hebben dan jou. Krijg namelijk de onderstaande output als ik om de help vraag.
Heb deze instructies gebruikt om m te installeren --> https://github.com/mhvis/solar/tree/v2

Heb daarmee de onderstaande versie gekregen.
> $ samil --version
> samil, version 2.0.0.post2

> $ samil mqtt --help
> 
> Usage: samil mqtt [OPTIONS]
> 
>   Publish inverter data to an MQTT broker.
> 
> Options:
>   -h, --host TEXT     MQTT broker hostname/IP address.  [default: localhost]
>   -p, --port INTEGER  MQTT broker port.  [default: 1883]
>   --client-id TEXT    Client ID used when connecting to the broker. If not
>                       provided, one will be randomly generated.
> 
>   --tls               Enable SSL/TLS support.
>   --username TEXT     MQTT username.
>   --password TEXT     MQTT password.
>   --interface TEXT    IP address of local network interface to bind to.
>   --help              Show this message and exit.

mhvis, 2020-12-17:
Ah dit klopt inderdaad, de gepubliceerde versie is iets ouder zie ik. Maar het zou niet uit moeten maken, als het verbinden met broker en inverter lukt wordt als het goed is de topicnaam geprint.

WendelVanRheenen, 2020-12-17:
opmerkelijk; hij blijft blanco bij mij... heb voor de zekerheid nog even mosquitto getest. Gezien t een speeltuin is op een raspberry pi om m'n huidige omgeving te vervangen. 

Verder ook even samil mqtt open laten staan om te zien of ie meer tijd nodig heeft. Maar er gebeurd ogenschijnlijk niks. Zie ook bijgevoegde afbeelding. Hierop zie je 2 raspberry pi's.

[image]

mhvis, 2020-12-17:
Je hebt gelijk, de gepubliceerde versie is inderdaad niet de juiste zie ik nu ([het is deze](https://github.com/mhvis/solar/blob/efc8f5ca86bd8b822b89ff2b2ea47f674e2f1767/samil/cli.py)). Ik heb de nieuwste en juiste versie alleen als pre-release gepubliceerd blijkbaar. Je kan de pre-release installeren met pip:

`$ pip3 install --user --upgrade --pre samil`

Bedankt voor het laten weten, ik zou het eigenlijk moeten opschonen en de nieuwste versie publiceren alleen omdat ik het script zelf niet meer gebruik komt het er niet van.

WendelVanRheenen, 2020-12-17:
Leuk dat je dat zo zegd, heb t zelfde gevoel maar dan voor het tegenovergestelde.  Ben blij dat je me zomaar wilt helpen, ondanks we elkaar niet kennen. 

Heb de update zojuist tussendoor gedaan. Maar hij leek t zelfde gedrag te geven. Heb nog niet met veel aandacht kunnen kijken. Morgen nog een keer proberen.

mhvis, 2020-12-17:
Het versienummer dat `samil --version` geeft zou 2.1.0a5 moeten zijn.

WendelVanRheenen, 2020-12-17:
> Het versienummer dat `samil --version` geeft zou 2.1.0a5 moeten zijn.

Ah, ondanks dat ie wat installeerde is de upgrade niet goed gegaan.  Heb nog steeds 2.0.0

mhvis, 2020-12-17:
Je kan controleren of de versie geïnstalleerd door pip juist is met `pip3 list --user`, maar ik weet zo even niet waar dit aan ligt.

WendelVanRheenen, 2020-12-18:
@mhvis  whoop whoop, goud dit! heb hm eerst geuninstalleerd en toen opnieuw geinstalleerd, kreeg weer v2.0.0 en die heb ik succesvol kunenn upgraden. Nu doet ie t! :+1: 
[image]

WendelVanRheenen, 2020-12-18:
> > Het versienummer dat `samil --version` geeft zou 2.1.0a5 moeten zijn.
> 
> Ah, ondanks dat ie wat installeerde is de upgrade niet goed gegaan. Heb nog steeds 2.0.0

heb idd nu versie 2.1.0a5

mhvis, 2021-01-08:
De nieuwste versie is nu correct gereleased.

closed 2021-01-08 by mhvis