mhvis/solar #19: SolarLake10K (Three-Phase) issue

alberthoekstra, 2021-01-03:
Hi,

I finally found this repo, and I'm so glad I did!
Thanks for the work you've done already and I hope you can help me out with this.

So I've got a Three-Phase inventer and this causes some issue's in the data read methods.

Samil version 2.1.0a5.

Model info:
```
Device type.............Three-phase inverter
VA rating...............10000
Firmware version........1.36
Model name..............SolarLake10K
Manufacturer............SamilPower
Communication version...1.36
Other version...........1.36
General.................1
```

Status data
```
Status data #1
Operation mode...........Normal
Total operation time.....25118 h
PV1 input power..........71 W
PV2 input power..........79 W
PV1 voltage..............442.8 V
PV2 voltage..............460.9 V
PV1 current..............0.1 A
PV2 current..............0.1 A
Output power.............90 W
Energy today.............5.00 kWh
Energy total.............57257.2 kWh
Grid voltage R-phase.....228.5 V
Grid current R-phase.....0.9 A
Grid frequency R-phase...49.97 Hz
Grid voltage S-phase.....229.6 V
Grid current S-phase.....1.0 A
Grid frequency S-phase...49.97 Hz
Grid voltage T-phase.....229.5 V
Grid current T-phase.....1.2 A
Grid frequency T-phase...49.97 Hz
Internal temperature.....35.0 °C
Heatsink temperature.....36.0 °C
```

Error:
```
samil pvoutput [REMOVED] [REMOVED]
INFO:root:Searching for inverters
INFO:root:Connected with inverter on address ('192.168.1.x', 1219)
Traceback (most recent call last):
  File "/home/albert/.local/bin/samil", line 8, in <module>
    sys.exit(cli())
  File "/home/albert/.local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/albert/.local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/albert/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/albert/.local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/albert/.local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/albert/.local/lib/python3.8/site-packages/samil/cli.py", line 291, in pvoutput
    upload()
  File "/home/albert/.local/lib/python3.8/site-packages/samil/cli.py", line 287, in upload
    voltage=sum(s["grid_voltage"] for s in statuses) / len(statuses))
  File "/home/albert/.local/lib/python3.8/site-packages/samil/cli.py", line 287, in <genexpr>
    voltage=sum(s["grid_voltage"] for s in statuses) / len(statuses))
KeyError: 'grid_voltage'
```

The error says enough I think, key 'grid_voltage' is not availble in this message.
For me the voltages of the differtent phases are not really interesting. So just picking a random voltage key would be enough.

Btw. I havent testen with MQTT yet, but maybe the MQTT functionality has the same issue.

I'm a C# developer myself, not really sure how I could help you to fix this.
Let me know if I can do anything for you please.

mhvis, 2021-01-07:
Hi, great to hear :smile:! I cleaned up the code yesterday and published a new release (2.1.0). Indeed I had not yet taken the voltage for 3-phase inverters into account. Now it takes the average of the 3 voltages, or if you specify `--dc-voltage` it will use the panel voltage instead, as average of PV1 voltage and PV2 voltage. If you prefer different behavior, e.g. using a random phase instead of taking the average, let me know, I can easily change it (the aggregation code is [here](https://github.com/mhvis/solar/blob/6114977b367e151f5b298c377c3d2c6f66ba4d7c/samil/pvoutput.py#L56)).

Here's the command to upgrade (assuming user installation using pip3):

```
$ pip3 install --user --upgrade samil
```

I have not tested thoroughly so you might still encounter bugs.

The MQTT code doesn't aggregate values and therefore shouldn't have the same issue.

alberthoekstra, 2021-01-10:
Works like a charm. Thanks!

closed 2021-01-10 by alberthoekstra

mhvis, 2021-01-10:
Top!