mhvis/solar #20: SolarRiver 3000 TL => can't find inverter via simal monitor

konehead01, 2022-05-21:
Hi,

Many thanks for you solution. I installed your solution today. I have an SolarRiver 3000 TL. My Raspberry  is in the same network as my Inverter. The Inverter IP is 192.168.1.53 (able to ping/traceroute it via the commandline).  The Samil monitor command say;s => adress not avaliable when I use the commando 'samil monitor'

I tried to change the Network Parameters setting in my inverter. I am able to select TCP Server, TCP Client or UDP. I am able to update Port ID, Server adress and time out. Which do I have to change? And Which IP do I have to select by server adress, the IP of my Raspberry? Hope you are able to help me. I am Dutch btw

mhvis, 2022-05-23:
I am not sure what the issue is, but it seems that in general there are often issues with the SolarRiver TL series. It might be that those series of inverters do not work well with this tool, as opposed to the other SolarRiver TL-D and SolarLake TL series. There is therefore a good chance that is just won't work, I'm sorry to say (I'll put a notice about this in the readme).

What you could try is to see if the original software from Samil Power works. This tool is based on that software, so if the original software does work, this tool should work as well. However it is a bit annoying, you have to get it from [this Chinese website](https://web.archive.org/web/20190720131520/http://www.samilpower.com/service.php), at the bottom, it is called "SolarPower Browser (V3) - Monitoring Software ONLY for SolarRiver TL,SolarRiver TL-D, SolarLake TL". If you're not willing to try this, I totally understand.

There is a small chance it is just a network issue. If you can copy or screenshot the exact command (`samil monitor`) and the output, that could rule this out.

I'm not sure about the Network Parameter settings, on my SolarRiver TL-D inverter I have them as the default settings. Server address could be the IP of your Raspberry indeed, Port ID is probably 1200.

konehead01, 2022-05-25:
Hi Maarten! 

Many thanks for your reply and sorry for the late response. Uhmm, i can't use the SolarPower Browser, since i have only Mac IOS software :(. Please find attached the requested print screen, i hope it is a network error. 

[image]

mhvis, 2022-05-25:
Try just `samil monitor` as command, without the `--interface` option (it is to force a specific network interface, but normally not necessary).

For network protocol you could try both TCP-Client and TCP-Server to see if it makes a difference.

konehead01, 2022-05-26:
hi, ahh, thanks! Did execute the command without the --. Got the following message -is not good I assume-:

[image]

mhvis, 2022-05-26:
Leave out `interface 192.168.1.53`, just type `samil monitor` (2 words) and hit enter. It is not necessary to give the inverter IP, it tries to find the inverter automatically.

mhvis, 2022-06-04:
Then I'm afraid it probably won't work with these inverter series, I will update the README.

The tool might need to be updated to make it work but as I don't have this inverter series I can't do this.