mhvis/solar #22: Crashes at night

timsmithlx, 2023-01-16:
I have just started using your software and thank you so much for the hard work

I am connection to the TL series. If the sun is out I can connect, but at night I can’t.

this causes the service to stop over night - is there a way round this? 

The screen on the inverter says “please wait…..” until the sun comes out? thanks in advance 

running on a Pi400

mhvis, 2023-01-17:
It might stop or crash overnight when the inverter is off. However, it should connect again when the inverter is back online.

You didn't provide details like which error you encountered, which commands you uses, which OS, inverter model et cetera. So I can't help you further at the moment.

timsmithlx, 2023-01-17:
Sorry I was very undescriptive!

I have now realised that is exactly how it works!

One question, I am integrating into a Homeassistant build using MQTT 

is it possible to have the mqtt service output “0” over night - currently I get “state unknown” on my HA integration… whacking all the charts off etc 

Thanks for this super handy tool

Running on. Pi as MQTT service (exactly as your tutorial)

mhvis, 2023-01-18:
> is it possible to have the mqtt service output “0” over night - currently I get “state unknown” on my HA integration… whacking all the charts off etc

That might be possible. Do you use the MQTT status messages? Those are JSON objects. Which fields do you use of them? Maybe `output_power`? It could be that those fields disappear during the night. I can change it so that they have 0 instead of disappearing.