mhvis/solar #26: Message 3 (04 02 00) decoded - Communications parameters

lwoggardner, 2024-04-05:
So more TCP dumping,  spotted my wifi password in the output....  which lead to decoding the 04 02 00 message
I'm working with a  SolarRiver TL-D device which reports versions as 'V1.30', and Solar Power Browser running under wine on linux.

Format of the 04 02 00 payload is
  * 1 char - installed comms module,  '1' = GPRS,  '2' = WIFI,  3 = 'No Module' [1]
  * 35 char WIFI SSID
  * 60 char WIFI Password [2]
  * 4 char GPRS PIN
  * 30 char GPRS Centre Number
  * 30 char GPRS Customer Number
  * 1 char - 'OK to Save'  - 0 is ok,  non-zero is false [3]
 
Total 161 bytes of payload

[1] - '1','2' enable the WIFI/GPRS input fields in Browser,  0 or >3  seems to display some kind of error code.
[2] - the wifi password shows as '*********' in the app,  but is communicated in the clear on the wire - which apart from being useless to customer,  means any passer-by with an ethernet cable can read the WIFI password!!

Messsage to set parameters coming out of SolarPower Browser seem to be all 05 0x.  If 'OK to Save' is non zero in the  04 02 payload, then these messages will not be sent.

Note Browser prompts for an input password -  'samilpower88' seems to be hard coded into the app in that if you get it wrong nothing is sent.  Not sure yet how the inverter responds to indicate success/failure of  the update.

Set WIFI 05 00 00,  Payload 35 char SSID,  60 char Password,
Set GPRS 05 01 00.  Payload 4 char GPRS PIN,  30 char GPRS Centre, 30 char GPRS Customer

If the source of https://mhvis.github.io/solar/ is here somewhere, I'm happy to send a PR to update this info.

Being able to set the WIFI info without needing SolarPower Brower would be very handy so if I work that out, I will update here.

mhvis, 2024-04-10:
Oh wow amazing! Very cool

I believe my inverter unit does not have GPRS or WIFI so I probably couldn't have figured this out. Very nice to add this info.

The source of the protocol doc is also in this repo, in a different branch: https://github.com/mhvis/solar/blob/gh-pages/index.md. It should be possible to open a PR against the branch. I can also make a start later if I have some time.

mhvis, 2024-04-11:
Interesting! Then we might have the wifi module indeed and completely missed it 😄 . That's nice, I'm going to check that later. We didn't really look into this because we did have an ethernet connection nearby.