mhvis/solar #28: Serial number empty

deosrc, 2024-05-07:
I usually run in mqtt mode and noticed that the mqtt topic wasn't being set correctly. It seems that the inverter serial number is not being received correctly resulting in an empty segment in the mqtt topic of `solar//status` (using a topic prefix of `solar`).

When I run in monitor mode, this is the output:

```text
Searching for inverter
Found inverter on address ('192.168.1.77', 4110)
Model info
Device type.............Single-phase inverter
VA rating...............4500
Firmware version........V1.90
Model name..............River 4500TL-D
Manufacturer............SamilPower
Serial number...........
Communication version...V1.90
Other version...........V1.90
General.................2
Status data #1
Operation mode.........Normal
Total operation time...257 h
PV1 input power........594 W
PV2 input power........169 W
PV1 voltage............278.3 V
PV2 voltage............179.1 V
PV1 current............2.1 A
PV2 current............0.9 A
Output power...........763 W
Energy today...........0.00 kWh
Energy total...........0.0 kWh
Grid voltage...........237.3 V
Grid current...........3.4 A
Grid frequency.........50.05 Hz
Internal temperature...36.5 °C
```

I am using v2.2.4 but am unsure when the issue first appeared.

deosrc, 2024-05-07:
Turned out to be a problem with the inverter. When I checked the inverter it had lost it's serial number, mac address, all history and even the time. There wasn't anything in the events list.

Isolating it from both the mains and solar panels, waiting for it to fully power down, and then reconnecting it seems to have sorted it.

closed 2024-05-07 by deosrc

mhvis, 2024-05-11:
Great that you solved it. Currently it always puts the serial number in the topic. This is done in order to support multiple inverters. But it would be a good change to omit the serial number when there is only one inverter. Then the full MQTT topic would also be configurable.

Could be a possible future enhancement.