mhvis/solar #29: samil op Debian 12?

alochem, 2025-01-08:
Ik gebruik al enkele jaren samil v2.1.0 op een Raspberry Pi 3. Samil wordt alleen gebruikt om gegevens van de omvormer TL3400-D naar PVOutput.org te sturen. De Pi3, waar ook enkele jaren Domoticz op heeft gedraaid, is echter vervangen door een Pi5 met Home Assistant. Naast de Pi5 heb ik nog een P4 draaien waar PiHole op is geïnstalleerd.

Nu dacht ik samil te verplaatsen van de Pi3 (PRETTY_NAME="Raspbian GNU/Linux 10 (buster)") naar de Pi4 (PRETTY_NAME="Debian GNU/Linux 12 (bookworm)") en daarbij gelijk op te waarderen naar  samil v2.2.3.

Het lukt me echter niet om samil (v2.1.0 of v2.2.3) aan de gang te krijgen op de Pi4. Wanneer ik de optie '--dry-run' mee stuur gaat het goed, laat ik die optie weg zie ik onderstaande verschijnen (ID en apikey verwijderd):

```
arno@pihole:/ $ /home/arno/.local/bin/samil pvoutput <ID> <apikey>
INFO:root:Connecting to inverter(s)
INFO:root:Connected with inverter on address ('192.168.1.195', 4422)
INFO:root:Uploading status data: {'energy_gen': 1480, 'power_gen': 63, 'temp': Decimal('19.4'), 'voltage': Decimal('230.0')}
Traceback (most recent call last):
  File "/home/arno/.local/bin/samil", line 10, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/samil/cli.py", line 256, in pvoutput
    upload()
  File "/home/arno/.local/lib/python3.11/site-packages/samil/cli.py", line 252, in upload
    add_status(system_id, api_key, **status_data)
  File "/home/arno/.local/lib/python3.11/site-packages/samil/pvoutput.py", line 53, in add_status
    return urlopen(req)
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 525, in open
    response = meth(req, response)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 634, in http_response
    response = self.parent.error(
               ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 563, in error
    return self._call_chain(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 496, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 401: Unauthorized
arno@pihole:/ $
```

Enig idee wat hier mis kan zijn?

mhvis, 2025-01-09:
De onderste regel geeft aan dat PVOutput een respons geeft van `HTTP Error 401: Unauthorized`. Kan het misschien zijn dat de inverter ID of API key niet kloppen?

Ik heb net nog wel de package structuur verbeterd in v2.3, maar dat zal waarschijnlijk geen verschil maken.

mhvis, 2025-01-09:
Goed om te horen! Veel plezier ermee.

alochem, 2025-01-09:
Hoi Maarten,

Ik heb toch ook even getest met v2.3.
Leek het in eerste instantie goed te werken, later blijkt dat er uploads
gemist worden:

[image: ac376c66-52a5-4904-b0ab-2b27e4ad1b96.png]

Je ziet dat bijvoorbeeld15:10, 15:2 en 15:35 missen. In de logging van
crontab zie ik dat de opdracht wel is uitgevoerd:

2025-01-09T15:05:01.896970+01:00 pihole CRON[464880]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:10:01.863051+01:00 pihole CRON[465008]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:15:01.361088+01:00 pihole CRON[465071]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:20:01.551183+01:00 pihole CRON[465264]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:25:01.767075+01:00 pihole CRON[465342]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:30:01.443359+01:00 pihole CRON[465439]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:35:01.198686+01:00 pihole CRON[465576]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)
2025-01-09T15:40:01.505869+01:00 pihole CRON[465707]: (arno) CMD
(/home/arno/.local/bin/samil pvoutput 34903
27b930b05e949625c16aa611a6056e013b09ee80)

Ik heb één keer bij het handmatig uitvoeren van de opdracht een foutmeling
zien verschijnen, maar uiteraard kan ik nu uitvoeren wat ik wil, het gaat
niet meer fout.
Zodra ik de foutmelding weer in beeld heb stuur ik 'm naar je toe.
Of wordt er toch ergens een logging bijgehouden?

Groet,
Arno

mhvis, 2025-01-09:
Thanks voor het testen. Ik heb zelf nog geen tijd gehad om de nieuwe versie thuis te testen. Dat doe ik binnenkort een keer.

Je zou de --interval optie kunnen proberen en dan het commando een tijdje op de voorgrond laten draaien. Dat maakt het waarschijnlijk makkelijker om de error te zien. Bijv. `samil pvoutput --interval 5 <system id> <api key>`. Als er een error komt stopt het script met uploaden.

alochem, 2025-01-10:
Hallo Maarten,

Vanmiddag de --interval optie aangezet, waarbij ik eerst de crontab even
uitgezet heb.
Het duurde niet lang of onderstaande melding kwam in beeld:

***@***.***:~ $ /home/arno/.local/bin/samil pvoutput --interval 5 <ID>
<API-key>
INFO:samil.cli:Connecting to inverter(s)
INFO:samil.inverter:Connected with inverter on address ('192.168.1.195',
4251)
INFO:samil.cli:Waiting for next interval boundary to do the first upload
Exception in thread Thread-1 (_ka_runner):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/arno/.local/lib/python3.11/site-packages/samil/inverter.py",
line 434, in _ka_runner
    self.keep_alive()
  File "/home/arno/.local/lib/python3.11/site-packages/samil/inverter.py",
line 442, in keep_alive
    super().receive()
  File "/home/arno/.local/lib/python3.11/site-packages/samil/inverter.py",
line 177, in receive
    return read_message(self.sock_file)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/arno/.local/lib/python3.11/site-packages/samil/inverter.py",
line 345, in read_message
    raise InverterEOFError
samil.inverter.InverterEOFError

Kun je hier wat mee?

alochem, 2025-01-18:
Hoi Maarten,

Even een vraagje over de geïnstalleerde service waar ik nu wat mee aan het
testen ben.
Om de één of andere manier kapt de service ermee na 1 keer uploaden:

Jan 18 12:20:27 PiHole systemd[1]: Started samil.service - Samil.
Jan 18 12:20:27 PiHole samil[1932]: INFO:samil.cli:Connecting to inverter(s)
Jan 18 12:20:28 PiHole samil[1932]: INFO:samil.inverter:Connected with
inverter on address ('192.168.1.195', 4109)
Jan 18 12:20:31 PiHole samil[1932]: INFO:samil.cli:Uploading status data:
{'energy_gen': 420, 'power_gen': 243, 'tem>
Jan 18 12:20:32 PiHole systemd[1]: samil.service: Deactivated successfully.
Jan 18 12:20:32 PiHole systemd[1]: samil.service: Consumed 1.044s CPU time.

Dat kan natuurlijk niet de bedoeling zijn.
Ik ben daarom wat in het bestand samil.service aan het hacken geslagen en
heb de restart op 'always' gezet met een Restartsec van 1s.
De service wordt nu weer elke keer gestart nadat 'ie afgesloten is, maar
dit lijkt me niet de juiste manier.

Ook de Environment-setting die in het bestand staat geeft een error:

Jan 18 11:59:58 PiHole systemd[1]: /etc/systemd/system/samil.service:17:
Invalid environment assignment, ignoring: is
Jan 18 11:59:58 PiHole systemd[1]: /etc/systemd/system/samil.service:17:
Invalid environment assignment, ignoring: as
Jan 18 11:59:58 PiHole systemd[1]: /etc/systemd/system/samil.service:17:
Invalid environment assignment, ignoring: Leave
Jan 18 11:59:58 PiHole systemd[1]: /etc/systemd/system/samil.service:17:
Invalid environment assignment, ignoring: #

Deze heb ik er vooralsnog even uitgehaald, dat lijkt goed te gaan.

Heb jij een idee hoe is die service/environment goed krijg?

mhvis, 2025-01-21:
Ha!

Je hebt neem ik aan het commando `samil pvoutput` gebruikt bij de optie ExecStart, zonder de --interval optie. In dat geval doet het script direct een upload en stopt daarna. Dit werkt niet goed samen met een standaard SystemD service, omdat die bedoeld zijn voor langdurige processen die niet uit zichzelf stoppen. Het PVOutput commando is zo gemaakt om samen te gebruiken met een task scheduler zoals cron. Cron staat standaard geïnstalleerd op Debian. Het enige wat je hoeft te doen is `crontab -e` om een job toe te voegen. Om het script elke 5 minuten aan te roepen gebruik je deze regel:

```
*/5 * * * *  /path/to/samil pvoutput <je system id> <je API key> 2>&1 | /usr/bin/logger -t samil
```

Als je niet zeker weet waar `samil` staat geïnstalleerd, het commando `which samil` vertelt je dat.

Het stuk vanaf `2>&1` is niet noodzakelijk maar zorgt ervoor dat je de output van het script kan teruglezen in de system logs, met iets als `journalctl -t samil` (uit mijn hoofd).

---

Je kan ook een SystemD service gebruiken, maar dan moet je het vertellen dat het een kortlopend proces is, met de optie `Type=oneshot` (google 'systemd oneshot'). Daarnaast moet je een timer aanmaken om de service elke 5 minuten te laten starten. Google daarvoor `systemd timer`. De exacte instructies kan ik je geven maar kost extra van mijn tijd.

Eventueel kun je de --interval optie gebruiken. Dan wordt het een langdurig proces en kun je de standaard SystemD service gebruiken zoals in de README. Maar deze optie is minder robuust.

---

De Environment-setting bevat zo te zien een fout. Het stuk `# Leave as is` moet eruit. Ik heb de README aangepast en verbeterd.

---

Om terug te komen op je vorige comment; die error kan veel oorzaken hebben. Mogelijk lost het zich vanzelf op. Als je continu InverterEOFError krijgt weet ik niet direct de oorzaak (EOF=end of file). Als ik het niet zelf kan reproduceren kan ik je niet helpen.

mhvis, 2025-01-21:
Sorry je gebruikte al cron voorheen. Ik had even niet ver genoeg teruggelezen.

Als je liever SystemD gebruikt zul je dus een timer moeten gebruiken of de --interval optie.