mhvis/solar #31: Home assistant HACS fork?

bgh91, 2025-02-23:
Can this be implemented into home assistant using HACS integration?

mhvis, 2025-03-02:
Yes it would be relatively straightforward to implement this in a HACS integration or even offer it up for inclusion in Home Assistant directly.

If someone writes this integration I would be more than happy to try it out in my own Home Assistant instance. I'm not going to create this integration myself voluntarily.

timmmmmmmmm, 2025-03-19:
Hey @mhvis,

I’ve turned your code into a HACS integration! This is my first Home Assistant integration, and I developed it heavily using the Cline/Claude VSCode plugin, so it’s still an early, bare-bones version. However, in my limited testing, it seems to work fine: https://github.com/timmmmmmmmm/ha_samil_power

I set it up as a separate repo to make it easier to add to HACS. Does this approach align with your vision for the project? Since the actual code is your work, I want to make sure I’m giving you proper credit and handling rights correctly. Let me know if there’s anything I can do to keep things in sync or acknowledge your contributions in a way you prefer.

Laat maar even weten :)

mhvis, 2025-03-19:
Heel gaaf! Ziet er nice uit!

Good idea to put this in a separate repository. You can tackle this any way you like. You don't have to give credit, but I do appreciate it 🙏😄. 

You do have to adhere to the license. Which means that the license notice should be available alongside the source code. I see that I didn't add a license header to each file, only a LICENSE file. I just corrected this in my repo. Can you make sure to copy the new source files over to your samil folder? (Or just copy the LICENSE file to your samil folder, that's fine as well.)

You actually shouldn't need to copy the files. You already have the PyPi package listed as a requirement in your manifest.json, so the samil package should be installed, and it should work without the copied samil directory. Then the license is also implicitly applied.

When I have time somewhere in the upcoming weeks I'll try the integration and get back to you. Thanks very much for creating this 😄!